chilipeppr/tinyg #13: AdBlock extension for Chrome seems to kill chilipeppr

Scalpel78, 2015-01-11:
Hi, 
I've got the Adblock Plus extension installed in Chrome. When navigating to chilipeppr.com I only get a blank page. The console says this:

[image]

If I add chilipeppr.com to the list of whitelisted domains for Adblock the page loads as expected.

It would be helpful if chilipeppr had shown an informational text saying it was unable to load, and that perhaps Adblock was the reason.

peteruithoven, 2015-02-03:
One tricky thing that it depends on what plugin you use. It also happens when using disconnect or uBlock. 
Maybe we can display something like: 
> If you see this you need to make sure a browser extension, like Adblock, Disconnect or uBlock isn't blocking requests to Google. These are necessary because this website uses the Google App Engine. 

Of course it would be better if we can do this without requests to Google, since privacy etc.

peteruithoven, 2015-02-03:
Also see: https://github.com/chilipeppr/tinyg/issues/3

chilipeppr, 2015-02-04:
I'm not sure it makes sense to play whackamole with ad blockers? If you
install an app that modifies how a browser works, it seems to me you do so
at your own risk.