chilipeppr/tinyg #16: Workpiece ZXY Corner "locator"

stylewarz, 2015-02-05:
Hello

I wanted to ask if it is possible to use the TinyG Controller with Chilipeppr to determine the  Z hight of a work piece as well as the corner XY of it.

I've found this https://github.com/synthetos/TinyG/wiki/Automatic-Z-Homing-When-Milling-PCBs

Would it also work with a offset like shown here on the Mach3 Software? -> http://www.themakersguide.com/home/products/triple-edge-finder-2

I guess this would be a very cool solution for a lot of people using TinyG

Greetings, Philip

PS: I'm very new with CNC and hope this is the right place to do so. If not, please tell me where to go as I have not found any information on this yet.

UltiBots, 2016-06-23:
+1 for this feature... Thanks Brad

chilipeppr, 2016-06-23:
Yes this is definitely possible. I think there was a post on the Google
plus forum a while ago. You could do it manually using the probe command
like you see in the touch plate widget. I would fork the touch plate widget
and add x and y operations to it since it is just a z operation.

UltiBots, 2016-06-23:
John... thanks for your quick reply, great software, and guidance on how to
add this feature. Hopefully someone with current coding skills will
implement this. Best - Brad

Brad Norman
UltiBots LLC
772-257-1000
9am - 7pm Eastern
www.UltiBots.com