chilipeppr/tinyg #17: No via drilling holes with eagle brd import

Cinezaster, 2015-02-09:
I'm experimenting with the eagle brd import. But can't get the drill commands working for the via's and through hole components

johnlauer, 2015-02-09:
Sadly those are not implemented yet. What's even sadder is that in the scope of how hard it was to get the project to it's current state, which I think is an amazing state, the super simple via and pad drill toolpaths aren't implemented yet and that's probably only another 2 hours of coding to get them completed. They're coming soon and truly that Eagle BRD link probably shouldn't have been put public for this reason, but I'd love to ask you how your board rendering looks otherwise? The Eagle BRD import has been tested on roughly 10 different board designs including the Ardiono Eagle BRDs and they look great except for some polygons not rendering 100% correctly with respect to overlays and keepouts.

Want to help contribute any Javascript code to help get this project further along? We need to get beyond PCB2Gcode because it's holding all of us back.

Cinezaster, 2015-02-10:
The board rendering looks fine. Only I had designed my board with all my pads on the bottom :-(. This didn't worked I had to mirror everything to have the pads being cut. 
A way to switch from one side to the other would be a nice option. 

I would like to contribute on the js code. Only I'm a bit confused about how this jsfiddle thing is working. I'm using git for all my projects (we have an own hosted gitlab server) at work.

An other thing about this project is who has the lead and keeps track of the roadmap. And I'm missing documentation. For instance I like the new touch plate feature but have no clue how to hook them up.

johnlauer, 2015-02-10:
Well, flipping the board is also a next step that is also probably only an hour of work. If you look at the code that's rendering the top layer it's just asking the methods to render layer "1" which is Eagle's reference for that. Asking to render layer "16" would give you the bottom but I just never put a UI element in there. I wanted to give folks an option to gang up the top and bottom side by side so you could mill them in one long job together and then just crazy glue both top and bottom together (which is how i'd prefer to do it rather than flipping and aligning).

I'm in charge of the whole ChiliPeppr project, thus roadmap and pull requests. There are lots of contributors.

JSFiddle takes some getting used to. It's not quite Github, but it helps folks see the individual widgets way easier and to fork pretty easily. If you wanted to use Github you could copy the JSFiddle to Github and then go from there.

The Touch Plate could use a tutorial, that's for sure.

Cinezaster, 2015-02-13:
Hi @johnlauer 

Have never collaborated on a jsfiddle project. 
I added support for the holes as a starter:
```javascript
EagleCanvas.prototype.parseHole = function (hole) {
	return {
		x: parseFloat(hole.getAttribute('x')),
		y: parseFloat(hole.getAttribute('y')),
		drill: parseFloat(hole.getAttribute('drill'))
	};
};
```
And in the EagleCanvas.prototype.parse I added
```javascript
this.holes = {};
var holes = this.boardXML.getElementsByTagName('hole');
for (var holeIdx = 0; holeIdx < holes.length; holeIdx++) {
	var holeDict = this.parseHole(holes[holeIdx]);
	this.holes[holeIdx]= holeDict;	

}
```

http://jsfiddle.net/cinezaster/do9cc5zh/74/ How can I do ask for a pull request.

chilipeppr, 2015-02-13:
Pull requests are manual here. I can do it but it would seem to make sense
to wait until you've got it working. If I'm already rendering vias and
pads, what are the holes you are parsing? The vias and pads would have a
drill diameter already figured out by the current codebase, which wasn't
easy since Eagle uses a lot of assumptions on drill sizes. It would seem
you would use those values and generate the drills from that.

Cinezaster, 2015-02-13:
Ok no problem. 

About the holes these are mounting holes and I found out about them by opening my .brd file. 
```
<hole x="2" y="-3" drill="3.2"/>
<hole x="49" y="-3.1" drill="3.2"/>
<hole x="2" y="26" drill="3.2"/>
<hole x="49" y="26" drill="3.2"/>
```
We could possibly mill them out instead of drilling. I think it is not that hard to add.