chilipeppr/tinyg #20: Simulation OK, but not cutting correctly

JimF42, 2015-03-13:
I have a simple Z-Axis knob that I am trying to cut. While the simulation works just fine, the actual cutting is all wrong. The first the g-code file does is to cut the center hole. When it tries and cut it, it goes to the center, cuts the first step-down of the circle, then goes back to the XY origin (without resetting Z), then goes back to the center to cut another step-down, back to the XY origin, etc., etc. etc. All the while dragging the end mill through the material as it goes back and forth. Oh, and it does this little "jig dance" every time it gets back to the XY origin--as if it is telling the other bees where the honey is at... I tried different things, but the only thing that appears correct is the simulation. 

I will be happy to send you the g-code file (generated by Cut 2D), if it helps you out.

JimF42, 2015-03-13:
I'm also running TinyG edge 440.14, which I needed to update to that fixed some power management bugs I reported.

chilipeppr, 2015-03-13:
Hi Jim, my gut tells me this is a TinyG firmware issue and you should post
on the TinyG forum instead of ChiliPeppr. We do work closely together, but
since the simulation is showing it fine, it's probably the firmware.

JimF42, 2015-03-13:
Hmm, OK, I will do that, but for further clarification, when I run (not simulate) the cut, the on-screen "end mill" is doing exactly what the physical CNC is doing. Are you saying it is doing that because the _status codes_ may be different?

chilipeppr, 2015-03-13:
When you're running your job, the 3D viewer is just showing the XYZ
coordinates that the TinyG sends back. So that makes sense that on-screen
is looking as whacky as what the TinyG is doing.

JimF42, 2015-03-13:
OK, good to know. I have opened the following ticket on TinyG
https://github.com/synthetos/TinyG/issues/131

JimF42, 2015-03-14:
OK, it turns out I was using 440.13 and .14. I flashed my TinyG with .14 and everything seems good now. Sorry for the confusion.

closed 2015-03-14 by JimF42