chilipeppr/tinyg #21: Cant connect to SPJS after restarting it

raulgotor, 2015-03-31:
I'm on Snow Leopard & Firefox. Some times chilipeppr becomes unresponsive. Normally, in that situation I disconnect from JSON and close the chilipepper's Firefox tab. Then, I relaunch both chilipeppr and JSON server but I get the following message at the OS X terminal:

Error trying to bind to port: listen tcp :8989: bind: address already in use, so exiting...logout

It seems that the port remains open and JSON cant connect anymore, but at chilipeppr JSON widget there is no JSON server connection...

I've tried at the terminal the command 

lsof -i TCP:8989

but it makes no improvement. Any clues?

chilipeppr, 2015-03-31:
You are getting that port blocked because the original SPJS is still
running. So I would not do what you're doing.

The bigger question is what do you mean by unresponsive? Do you mean you're
running a job and the browser seems to get so bogged down your CPU is at
100% and things seem unresponsive? If so, I would disable the 3d viewer
when running a job. You can disable in the pulldown menu.

Also, make sure you're using the latest SPJS of v1.8. It has the best
memory management. I noticed you posted over at UGS about after 20,000
lines UGS seems to pause. You are running into Java garbage collection
issues. SPJS is written in Go and has a different approach to garbage
collection so you should find that it stays fairly smooth through the whole
job.

If I haven't correctly guessed at what you mean by unresponsive could you
send over some screenshots and/or video of the issue?

raulgotor, 2015-03-31:
Hi,
Thanks for your answer. I've SPJS v1.8.
By unresponsive I mean that the arduino/Grbl doesn't seems to receive the commands. The 3D viewer is working smooth.  All controls in the chilipeppr interface are working. For instance, by clicking the Homing button, the $H command will be displayed in the serial console. Still, the machine won't respond.

How can I stop SPJS from running? Going to localhost:8989 shows nothing, and the only way I've found until now is restarting the computer.

Thanks in advance

raulgotor, 2015-03-31:
Hi,

I've got more clues. I've been trying to produce a laser raster image which is plenty of Gcode commands. Around line 30.000 machine started to produce small pauses (same as in the UGS post that you have seen). They were more and more frequent until machine stopped completely. At this point, pause, stop, reset commands won't do anything, so I had to power off the arduino, and after turning it on, it was responding normally again.

Then, I have been playing with the X/Y arrows, moving the machine to the left and to the right, and suddently it stop moving, and no command made a movement anymore. Then, I've realized that those not performed commands are queued as stated by the serial port monitor.    
These last queued commands are being executed now very very slow, more or less one every 3-5 minutes. I'm starting to think that maybe I have some hardware problem with the ATMega IC?

Here is the log (last 40 lines):

{"Version" : "1.81"}
{"Commands" : ["list", "open [portName] [baud] [bufferAlgorithm (optional)]", "send [portName] [cmd]", "sendnobuf [portName] [cmd]", "close [portName]", "bufferalgorithms", "baudrates", "restart", "exit"]}
{"Hostname" : "MacBook-2.local"}
send /dev/tty.usbmodem3d11 ~
{"Cmd":"Queued","QCnt":84670,"Ids":[""],"D":["~\n"],"Port":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":84669,"Id":"","P":"/dev/tty.usbmodem3d11"}
{"Cmd":"CompleteFake","Id":"","P":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":84668,"Id":"g86745","P":"/dev/tty.usbmodem3d11"}
send /dev/tty.usbmodem3d11 
{"Cmd":"WipedQueue","QCnt":1,"Port":"/dev/tty.usbmodem3d11"}
{"Cmd":"Queued","QCnt":2,"Ids":[""],"D":["\u0018\n"],"Port":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":1,"Id":"","P":"/dev/tty.usbmodem3d11"}
{"Cmd":"CompleteFake","Id":"","P":"/dev/tty.usbmodem3d11"}
send /dev/tty.usbmodem3d11 G91 G0 X1 G90
{"Cmd":"Queued","QCnt":2,"Ids":["",""],"D":["G91G0X1\n","G90\n"],"Port":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":1,"Id":"","P":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":0,"Id":"","P":"/dev/tty.usbmodem3d11"}
send /dev/tty.usbmodem3d11 G91 G0 X1 G90
{"Cmd":"Queued","QCnt":2,"Ids":["",""],"D":["G91G0X1\n","G90\n"],"Port":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":1,"Id":"","P":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":0,"Id":"","P":"/dev/tty.usbmodem3d11"}
Error writing to /dev/tty.usbmodem3d11 write /dev/tty.usbmodem3d11: device not configured Closing port.
list
{ "SerialPorts": [ { "Name": "/dev/tty.usbmodem3d11", "Friendly": "/dev/tty.usbmodem3d11", "SerialNumber": "", "DeviceClass": "", "IsOpen": true, "IsPrimary": true, "RelatedNames": null, "Baud": 115200, "RtsOn": true, "DtrOn": false, "BufferAlgorithm": "grbl", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.BlackBerry9300-BlackBer", "Friendly": "tty.BlackBerry9300-BlackBer", "SerialNumber": "", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "RelatedNames": null, "Baud": 0, "RtsOn": false, "DtrOn": false, "BufferAlgorithm": "", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.Bluetooth-Modem", "Friendly": "tty.Bluetooth-Modem", "SerialNumber": "", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "RelatedNames": null, "Baud": 0, "RtsOn": false, "DtrOn": false, "BufferAlgorithm": "", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.Bluetooth-PDA-Sync", "Friendly": "tty.Bluetooth-PDA-Sync", "SerialNumber": "", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "RelatedNames": null, "Baud": 0, "RtsOn": false, "DtrOn": false, "BufferAlgorithm": "", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.Nokia5800-DataTransfer", "Friendly": "tty.Nokia5800-DataTransfer", "SerialNumber": "", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "RelatedNames": null, "Baud": 0, "RtsOn": false, "DtrOn": false, "BufferAlgorithm": "", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.PuertoSerie", "Friendly": "tty.PuertoSerie", "SerialNumber": "", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "RelatedNames": null, "Baud": 0, "RtsOn": false, "DtrOn": false, "BufferAlgorithm": "", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.PuertoSerie-1", "Friendly": "tty.PuertoSerie-1", "SerialNumber": "", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "RelatedNames": null, "Baud": 0, "RtsOn": false, "DtrOn": false, "BufferAlgorithm": "", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.iMacdeMariCarmen-Blueto", "Friendly": "tty.iMacdeMariCarmen-Blueto", "SerialNumber": "", "DeviceClass": "", "IsOpen": false, "IsPrimary": false, "RelatedNames": null, "Baud": 0, "RtsOn": false, "DtrOn": false, "BufferAlgorithm": "", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 } ] }
list
{ "SerialPorts": [ { "Name": "/dev/tty.usbmodem3d11", "Friendly": "/dev/tty.usbmodem3d11", "SerialNumber": "", "DeviceClass": "", "IsOpen": true, "IsPrimary": true, "RelatedNames": null, "Baud": 115200, "RtsOn": true, "DtrOn": false, "BufferAlgorithm": "grbl", "AvailableBufferAlgorithms": [ "default", "tinyg", "dummypause", "grbl" ], "Ver": 1.81 }, { "Name": "/dev/tty.BlackBerry9300-Bla [... cut ...]
{"Cmd":"Write","QCnt":1,"Id":"","P":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":0,"Id":"","P":"/dev/tty.usbmodem3d11"}
send /dev/tty.usbmodem3d11 G91 G0 X10 G90
{"Cmd":"Queued","QCnt":2,"Ids":["",""],"D":["G91G0X10\n","G90\n"],"Port":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":1,"Id":"","P":"/dev/tty.usbmodem3d11"}
{"Cmd":"Write","QCnt":0,"Id":"","P":"/dev/tty.usbmodem3d11"}
send /dev/tty.usbmodem3d11 G91 G0 X10 G90
{"Cmd":"Queued","QCnt":2,"Ids":["",""],"D":["G91G0X10\n","G90\n"],"Port":"/dev/tty.usbmodem3d11"}

chilipeppr, 2015-03-31:
What version of SPJS are you using?

raulgotor, 2015-03-31:
v1.8

chilipeppr, 2015-03-31:
Can you send over screenshots and/or video of when things go wrong? I just
don't have enough info. My take is something is going wrong that you aren't
realizing that will only come out in a video or screenshots.

raulgotor, 2015-04-02:
Sorry for the delay. I was going to record a video, when I've realized that after changing and rerouting the USB cable this random behavior has gone away. It seems that it was related to EMI with the AC line where the air assist motor of the laser is connected. Sorry for the trouble, I've should have checked this first!