chilipeppr/tinyg #23: Auto-probe doesn't work with TinyG2 hardware

karlbackstrom, 2015-05-18:
Hi,

I have a Arduino Due with a gShield attached and running edge branch of the g2 firmware. In Chilipeppr the auto-level function was hanging on the first probe. 
The reason for the hanging is that the reply has changed between TinyG and TinyG2. The "prb" entry in the reply is now at top level. After changing the reply checking in the widget the auto-level is now working.

Just my 2 cents if other people have the same problem.

Line 1344:
```javascript
if ('prb' in json) { // Changed line
   // we hit the probe bottom. awesome. done.
   chilipeppr.unsubscribe("/com-chilipeppr-widget-serialport/recvline", that, callback);

   console.log("got to end of G38.2 cmd");
   probe.z = json.prb.z; // Changed line
                    
   probe.done = true;
```

chilipeppr, 2015-05-18:
Good catch. I bet we could look for both and cover both versions.

karlbackstrom, 2015-05-18:
Indeed, thanks for a very nice program.

chilipeppr, 2015-05-18:
Thanks for the bug fix!