chilipeppr/tinyg #25: Provide chrome extension to access local serial ports

matthijskooijman, 2015-06-23:
Right now, chilipeppr uses serial-port-json-server to access serial ports. For local serial ports, an alternative could be a chrome extension, which is essentially some javascript installed into the Chrome browser as an app/extension (not sure what the right terminology is there). Inside such an extension, there is [an API available](https://developer.chrome.com/apps/serial) that allows accessing local serial ports.

Using such an extension could make it easier for people to get started with chilipeppr.

For an example on how to use this, see https://github.com/Pinoccio/chrome-app-pinoccio (used to upload sketches to Pinoccio / Arduino boards) or https://github.com/garrows/browser-serialport (library around the API, but I'm not entirely sure what that library adds, probably compatibility with the node serialport library or something).

johnlauer, 2015-06-23:
I explored this option at the beginning of the ChiliPeppr project and abandoned that approach because I thought it made for a more difficult experience because you can't access serial ports in a normal browser window. Instead, you have to install a Chrome app with a manifest to gain access. Those extra steps are worse than downloading a file and launching it. That approach also attaches you directly to Chrome rather than being browser agnostic. That approach also meant that you couldn't remotely connect to your serial ports off of a Raspberry Pi while running your fancy browser on your local laptop. So, with all of those aspects is why I decided to go with a standalone compiled app.

One of the further reasons that it was lucky I went with a compiled app is the contention that any gcode sender runs into with respect to byte counting and getting back responses from the controller. Since these controllers have barely any serial port buffer, it's easy for characters to fall off the buffer. The problem is the data coming back. If your software slows down for any reason and can't read the responses quickly enough coming back from the cnc controller, you lose that data forever. So, the garbage collection of your sending app or even the single-threaded nature of Javascript can become massive issues. To run in a clean standalone compiled fast app where you have almost perfect control of the app running is the answer to a real solution.

matthijskooijman, 2015-06-24:
I wouldn't agree that downloading and running a program is necessarily easier - clicking an install button for a Chrome app and never having to think about it again would seem useful to me.

I agree that having a standalone program is certainly useful, I was just suggesting this as an alternative for local serial ports only.

Good points about buffering and speed, I suppose that SPJS does a bit more than just forwarding bytes right now? I could imagine approaches where the speed of the forwarding app don't matter much, but I'm not sure how the buffering and reporting works in grbl and tinyg, so can't comment on that.