chilipeppr/tinyg #26: Runaway cutting with tinyg 440.16

wephotography, 2015-07-26:
Ive had my ox cnc machine running a few months now. Have run a dozen or so different pieces with no issues. With the update of chillipeppr to the new multi axes system i started running into problems with the maching not keeping proper zero points. So i updated to 440.14 and that fixed the zero out problem but i started getting runaway cuts at random points that do not show up in the gcode or in simulation. On some of my fonts it would cut a perfect circle starting on one of the letters and then return to what it was doing. It would continue to do that every time as if it was programmed to do so on every pass. I made a short video of what i am running into now. Might be worth noting i create my designs in photoshop, then import to inkscape and trace bitmap, then to makercam for the tool paths. Has always worked fine until now. If anyone has any idea where to start with this it would be much appreciated. Link to the video below.  

https://www.youtube.com/watch?v=QoZ0zM9lOtI

wephotography, 2015-07-26:
https://www.dropbox.com/sh/3xyeh29fyve7nyo/AADwyYdcGdh6QLqcd0xVgsmia?dl=0

link to my dropbox and g code files. first one is the file from the video, second file runs a circle off of the lower right corner of the E in eddins

chilipeppr, 2015-07-26:
I'm hearing that some folks are having issues with G2/G3 arcs still. Is
that where you see the problems?

wephotography, 2015-07-26:
its possible that could be it. Im pretty new so what is an arc? Sorry thats probably a dumb question and im researching and answer now

chilipeppr, 2015-07-26:
ChiliPeppr shows G2/G3 arcs as light cyan blue color. Straight lines, which
are G1 moves, are dark blue. The reason I show them different colors is
that G2/G3 arcs are more complex than a straight line. Ultimately the CNC
controller has to render them to straight moves and sometimes that
rendering has been buggy.

wephotography, 2015-07-26:
Looking at the code where it went crazy it was in fact on G2 and G3 where it went wrong 

Looking at my other file I know went crazy to check if thats where it wrong too

wephotography, 2015-07-26:
[image]

Sure enough that was where it went wrong that time. I used that font on another cut and it did the same thing but on a different letter. I guess i should move this to the tinyg support forum?

chilipeppr, 2015-07-26:
Yup.

wephotography, 2015-07-26:
Ok thanks for the time and helping a noob identify the problem so quick.