chilipeppr/tinyg #28: Communication delays

stg, 2015-08-27:
System is Windows 7 Ultimate - x86.

Chilipeppr recently "joined" the two serial ports exposed by TinyG.

After this change, the first thing I noticed was that manual commands to move the machine did not take immediately.
Instead they appear as queued on the serial port (bottom, next to connected port), but not on the TinyG queue (TinyG buffer under feedhold/machine control).
Sometimes motions are executed immediately but mostly there is a delay of up to several seconds.
Often the queue just gets stuck and never reaches the board at all until i press "feedhold" followed by "resume". The queue is then dumped onto the board and all queued commands execute immediately.

Also, I've noticed that the (combined) port seems to disconnect at random - suddenly I notice there is no check in the checkbox for the port and I have to click it again to resume work. This may happen in the middle of a job.

Apart from this infrequent disconnect, g-code programs execute well and without any delays.

None of this has ever happened before and there has been no changes made to the system apart from the updates to Chilipeppr.

chilipeppr, 2015-08-27:
I've been doing a lot of testing lately and I think it's better to stick
with SPJS 1.80 for now. 1.83 is the source of the problems you're seeing.
1.84 will be released soon with fixes address this.

-John

stg, 2015-08-27:
Thank you!
I will try 1.80 next time I get a chance.