chilipeppr/tinyg #30: chilipeppr.com site down?

jedahan, 2015-09-30:
Hey, not sure where the appropriate place to report this is, but does the site require having a tinyg board plugged in for it to show anything? Could there be a "hey, plug in a tinyg for this site to work!" warning. Or a redirect to the project wiki or something?

chilipeppr, 2015-09-30:
The site does not require a tinyg to be plugged in at all for it to load. I
just tried the website just now and it's fine. It's hosted on Google App
Engine so it's never down other than some past times where it was over
quota, but then it would tell you it's over quota.

jedahan, 2015-09-30:
Tried on chrome and firefox on osx, I get "chilipeppr is not defined"....

[image]

jedahan, 2015-09-30:
[image]

peteruithoven, 2015-09-30:
Looks like a google related issue? Are you sure you don't have any add blockers / privacy enhancing browser extensions?

chilipeppr, 2015-09-30:
I just tried again to load the site and it's fine. Why don't you do a hard
reload. Right click on the reload button and choose "Hard Reload".

jedahan, 2015-09-30:
No privacy extensions, but I have google-analytics.com blocked in my hosts file.
Ublocked it, site works. Any way to have the site still work without google-analytics?

chilipeppr, 2015-09-30:
It would seem to me if you block that Javascript file that the site would
still load fine and work, so I'm not sure why you blocking that is causing
it not to load.