chilipeppr/tinyg #32: Gcode does not always load correctly

matthijskooijman, 2015-10-28:
While using the grbl workspace this afternoon, I found that a new gcode file is not always loaded correctly. I initially observed this while milling, when loading a new file after finishing another. Later, I reproduced this even when just switching between files in a clean grbl workspace, without any serial port attached or any milling happening (this did require a switching between files a few times, it usually worked). When the file was not loaded correctly, I think the 3D viewer did change (I'm not sure if it actually showed things correctly, though), but the contents of the Gcode panel did not change. I also think that the "File loaded" dialog was not displayed in this case, but I'm not entirely sure.

I tried reproducing this in the tinyg workspace just now, without any luck. However, I couldn't reproduce this in the grbl workspace again just now, so this might end up being a tricky problem to debug...

chilipeppr, 2015-10-29:
I've heard about this from other folks before and what was happening was an
error was occurring in the 3D viewer (even though it was showing the new
gcode correctly) when loading the file and thus step 2, which is the gcode
widget loading was never getting the event because the earlier event was
cancelled. So, could you perhaps open your javascript console and see if
any errors are occurring on load when you see this? Or could you send the
gcode file over and i can take a look? If it's truly random then that is
odd as well, but what it could be is file load order which tends to be
random based on when the browser loaded what javascript file. So seeing
javascript console errors can help debug on this  as well. My best bet is
there's a problem parsing the gcode file.

matthijskooijman, 2015-11-27:
I had another go at reproducing this today. Your comment made me think that, on a given page load, the problem should either occur all the time (possibly when combined with a specific gcode file), or not at all. Armed with that, I tried reloading the page and then switch between two gcode files, but after a dozen or two reloads, I still haven't managed to reproduce. No suspicious messages in the javascript console either. Perhaps this was unknowingly fixed, or some change was made that reduces the chances of this problem occurring. 

The gcode files we were working with that day are available here, though I don't think there's anything weird in them: https://gist.github.com/matthijskooijman/be29be54934bbef1b92f

johnlauer, 2015-11-27:
This got fixed a couple weeks ago in grbl workspace and about 4 weeks ago in TinyG workspace

matthijskooijman, 2015-11-27:
Cool! Let's close this issue then, I'll reopen if I run into this problem again.

closed 2015-11-27 by matthijskooijman

kubarbaz, 2019-12-13:
I think this problem still exists on jpadie workspace. I've experienced this when i'm trying to manipulate (auto leveled) z depth.