chilipeppr/tinyg #33: 3D viewer does not handle omitting G00/G01 properly

matthijskooijman, 2015-12-16:
AFAIU, the G00 and G01 commands are modal, meaning you can specify either of them once and then omit them later. E.g.:
```
G01
X0Y0
X10Y10
```
Should perform two G01 moves. This is also handled properly by the 3D viewer. However, when another command is inserted in between, like:

```
G01
G04 P0 (dwell)
X0Y0
X10Y10
```

The 3D viewer no longer shows two G01 moves but (none, in this case). I suspect that the modality of G00 and G01 isn't properly implemented in the 3D viewer, but instead it just remembers the last Gxx command for when it is omitted (or something along those lines, I didn't check the code).

Here's a (reduced) real-world example showing this problem, generated by pcb2gcode and simplified by me:

```
G94 ( Millimeters per minute feed rate. )
G21 ( Units == Millimeters. )

G90 ( Absolute coordinates. )
G00 Z2.00000 ( retract )

G00 X0 Y0 ( rapid move to begin. )
G01 Z-1 F100 (plunge)
G04 P0 ( dwell for no time -- G64 should not smooth over this point )

X10 Y0
X10 Y10
X0 Y10
X0 Y0

G00 Z20.000 ( retract )
```

chilipeppr, 2015-12-16:
Hmm. Yes, you are correct that the 3D viewer just keeps the last G command.
If you're up for tweaking the 3d viewer code that would be great. I don't
think I could get to fixing this bug anytime soon as it doesn't seem too
common with how Gcode is generated.

Since you're using pcb2gcode you could optionally use ChiliPeppr's Eagle
BRD Import. You just drag in your BRD file to ChiliPeppr.

matthijskooijman, 2015-12-16:
Hm, I might have a plunge then, though time is also limited for me. Shouldn't be a terribly complicated change, I would expect though.

> Since you're using pcb2gcode you could optionally use ChiliPeppr's Eagle
BRD Import. You just drag in your BRD file to ChiliPeppr.

I'm generating gerbers using KiCad, so an Eagle import wouldn't help me I think (perhaps KiCad has an Eagle export, but I don't thinks so).

chilipeppr, 2015-12-16:
Ok, well, perhaps try Flatcam then as i have not heard of the gcode having
that modal issue from flatcam.