chilipeppr/tinyg #34: grbl workspace will not load from jsfiddle directly (fix included)

matthijskooijman, 2015-12-16:


matthijskooijman, 2015-12-16:
W00ps, pressed enter too soon. Description is edited in now.

@jarretluft, I think this one is for you :-)

jarretluft, 2015-12-16:
@matthijskooijman Thanks for the fix, this has been updated.

matthijskooijman, 2015-12-16:
@jarretluft One more small thing: the url inside http://jsfiddle.net/matthijskooijman/yehnvsdp/ still points to the old fiddle, making the "fork this widget" button behave very much unexpectedly. Took me a while to figure out why the code in the fiddle wasn't the same as in my browser :-)

I just appended this here, since it didn't seem to warrant an entire new issue. Is this even the best channel for these kind of things?

matthijskooijman, 2015-12-16:
w00ps, had a wrong url in my comment, fixed now (http://jsfiddle.net/jarret/9aaL8jg4/)

matthijskooijman, 2015-12-17:
I retitled this issue to be a bit more generic, since I found some more things while going over the grbl workspace source.

Why does the grbl widget send a $G status command every 2s? It seems that nothing in the reply could change, unless a gcode command was executed. Wouldn't it be better to only send a $G command whenever the buffer becomes empty? Or perhaps send it every couple of seconds while gcode is being executed (to update status info during gcode runs too), and once after the queue empties?

The grbl workspace hides status commands and replies by overriding the send/receive functions of the serial port widget. There is a verbose button in the grbl widget to override this. However, the serial port widget already has a filter feature, with its own button, that filters out "ok" responses. Why not use this filter feature for all filtering?

jarretluft, 2015-12-18:
$G statuses are difficult because no solution will work in every case... I
settled on 2 sec interval when gcode is not running (the $G cycle stops
when gcode is executing), because this would capture the majority of
user-entered commands without having to create regex compares on every
command being sent on the /send or /jsonSend pubsubs.

Your suggestion of watching for an empty queue won't work on individual
commands because the SPJS queue would not increment until 254 characters
worth of gcode has been loaded into grbl's internal buffer.  Try sending a
G20 or G21 and notice that the queue count never increases or decreases...
This would not trigger a call for $G and the interface would not update to
let you know the new units that grbl is set to.  Same applies for WCS
changes.
Perhaps the SPJS grbl buffer could be adapted to do something like this,
but as I mentioned in another post, there is the possibility that push
responses are coming in GRBL 1.0b, so I would hold off making any changes
to this until we see what other options we have.

'ok' responses are basically useless. I filter them out separately so that
even when a user is looking in verbose mode, they don't have to filter
through endless ok responses.

matthijskooijman, 2015-12-18:
> Your suggestion of watching for an empty queue won't work on individual commands because the SPJS queue would not increment until 254 characters worth of gcode has been loaded into grbl's internal buffer. 

I was thinking about the buffer inside chillipeppr, not SPJS. However, you're right that having grbl push things is definitely a better solution.

> 'ok' responses are basically useless. I filter them out separately so that even when a user is looking in verbose mode, they don't have to filter through endless ok responses.

I guess a cleaner solution would be to have the serialport widget offer multiple levels of filtering, keeping the filtering controls in one place. But I see now why you added the separate verbose control, right.

Did you see my comment about the wrong url in http://jsfiddle.net/jarret/9aaL8jg4/ as well?