chilipeppr/tinyg #37: Jogging with keys stops spindle

lutorm, 2016-01-03:
It seems the way jogging is implemented is to send a move command when the key is depressed and then a feedhold when it is released. The problem with this is that the feedhold also stops the spindle, so it's impossible to use an edge finder. Using the buttons in the gui do not stop the spindle, but it's a big pain to move with them since you have to click repeatedly.

chilipeppr, 2016-01-03:
Interesting. I've never heard of anybody doing edge finding with their
spindle turned on. How does that work?

You'd have to ask the Synthetos guys on a way to not have % stop the
spindle. That's not really a ChiliPeppr issue.

One idea would be for you to add a toggle option to the Axes widget if you
fork it. Then let the user toggle off the sending of the feed hold and
buffer wipe. The jogging is done by sending lots of tiny movements, so to a
large degree those movements are close to when you lift your fingers off
the keyboard as long as you aren't sending long moves.

What's interesting is i've been thinking about improving jogging by always
issuing a long 1000mm move and then 100% relying on feed hold / buffer wipe.

You up for forking the widget and trying to see if you can get it to work
without buffer wipe?

lutorm, 2016-01-04:
Using an edge finder requires the spindle to spin, as in
https://youtu.be/f0od-cp_9dg?t=50s

If you issue a bunch of tiny moves, you'd have to rate the commands so
they're consumed at the same rate you're sending them, otherwise they'll
fill up in the serial buffer, right? So if you're jogging at, say,
5000mm/min, you'd need to issue a 1mm move at a rate of 5000Hz. Any lower,
the buffer is starved and the move will stutter, any faster and it will
buffer and then it'll keep moving when you release.

I've never used Javascript so I'm not sure how steep the learning curve for
hacking the widget is.

I have an idea for jogging that I think would work better, but it involves
adding a custom command and a new machine state. Basically, you need to be
able to command the machine in "velocity mode" so that it just keeps going
at the specified velocity in the absence of new commands. That way you'd
only need a new command when the velocity changes, and you cancel the job
by commanding zero velocity. That requires a bunch of changes to TinyG
though. I've fixed a bunch of bugs in TinyG lately, but a change like that
would be a lot more complicated.