chilipeppr/tinyg #38: Riley from TinyG confirms that the gcode sender can corrupt the TinyG board

dgtlmoon, 2016-05-03:
Hi

After spending a lot of time on https://www.synthetos.com/topics/yet-another-spindle-flashing-light-problem-why-is-it-so-easy-to-crash-the-tinyg/#post-9646 eventually I got a reply via email from Riley who said

"This is a newer issue where chilipepper is sending files that we cannot control.  Long story short this issue is not the same as the initial blinking issues.  However it does do the same thing.  We will fix this for you if you would like.
"

Essentially I accidentally sent some binary data in the gcode chillipeppr box and now my unit is requiring a reflash of the bootloader

What's frustrating is that this sounds like Riley Porter/TinyG already knew of the issue

chilipeppr, 2016-05-03:
I'm a bit confused how you were able to send binary data via ChiliPeppr? If
you only pasted in binary data, did you then also hit the play button to
send it? I read your forum post and it's not clear to me what binary data
you would have sent? When Gcode is read in from a file drag drop or a paste
operation it tries to 3D render it and place it in the Gcode widget where
it parses line by line. If everything looked completely wrong it would seem
to me you never would have hit play. If you never hit play, you never sent
your Gcode and thus there's no way that effected your device.

So, can you help shed light on this?

dgtlmoon, 2016-05-03:
It was a late night and I pressed play :) I accidentally dragged a zipped SVG file into the area and pressed play

dgtlmoon, 2016-05-07:
Ok so I got mt mkII programmer and reflashed the board and everything seems OK.

But, check this out, the gcode box contains SVG data half way through rendering...

[image]

chilipeppr, 2016-05-08:
Hmm. You might have found a bug. The SVG widget should stop the SVG file
from propagating to the Gcode widget itself.

dgtlmoon, 2016-05-08:
Yes something is not quite right with the SVG Import and the GCode component
[image]

I'm getting a few JS errors now also, will try to figure out how to reproduce it

dgtlmoon, 2016-05-08:
Hm, so Ive cleared all the stored SVG imports (drag and dropped to the workspace) in Chrome > Resources under 'local storage', reloaded the page, dragged a new SVG to the workspace, sent gcode to workspace and...

[image]

chilipeppr, 2016-05-08:
Can you post a screenshot?