chilipeppr/tinyg #40: Website broken, VM1343:1797 Uncaught TypeError: Cannot read property 'toFixed' of undefined

dgtlmoon, 2016-05-13:
Site seems totally broken, I'm unable to use, there is no 3D view.

my localStorage and SessionStorage is empty, I have tried deleting all cookies, logging out and logging in again

```
VM1343:1797 Uncaught TypeError: Cannot read property 'toFixed' of undefineddecorateExtents @ VM1343:1797
drawExtentsLabels @ VM1343:2114
drawAxesToolAndExtents @ VM1343:1909
openGCodeFromText @ VM1343:1110
init @ VM1343:149 
(anonymous function) @ VM1330:1027
context.execCb @ require.js:1650
Module.check @ require.js:866
(anonymous function) @ require.js:1113
(anonymous function) @ require.js:132
(anonymous function) @ require.js:1156
each @ require.js:57
Module.emit @ require.js:1155
Module.check @ require.js:917
(anonymous function) @ require.js
:1113(anonymous function) @ require.js:132
(anonymous function) @ require.js:
1156each @ require.js:57
Module.emit @ require.js:1155
Module.check @ require.js:917
Module.enable @ require.js:1143
Module.init @ require.js:774
callGetModule @ require.js:1170
context.completeLoad @ require.js:1564
context.onScriptLoad @ require.js:1671
VM1332:1756 serial port widget onPortList
```

dgtlmoon, 2016-05-13:
Update: seems to work with 'incognito mode' tho :(

dgtlmoon, 2016-05-13:
Update: tried in incognito mode, I see the default Chillipepr SVG import works fine, as soon as I drag and drop my own SVG (just a small rectangle 20x100) the whole site falls over

Attached is a screenshot of the console output and the SVG I drag+dropped over after doing a new incognito mode browser, and logging in via google. Note - github wont let me attach a .svg file so I've zipped it purely for attaching to this issue
[20x100.zip](https://github.com/chilipeppr/tinyg/files/263276/20x100.zip)

[image]

dgtlmoon, 2016-05-13:
Note - the test Chillipepr SVG works fine, something up with imported SVGs

chilipeppr, 2016-05-13:
I just tried your file and get same results. Why don't you try converting
the rectangle to a Path. The SVG Import is not super fancy and if you'd
like to extend it by all means I'd love some pull requests.

jsphpl, 2018-11-02:
Also getting `Cannot read property 'toFixed' of undefined` when trying to open the Example Chilipeppr SVG from the menu

jsphpl, 2018-11-02:
Update: It worked on the tinyg workspace, but not on grbl or jpadie

chilipeppr, 2018-11-02:
Hmm. I have not tried opening an SVG in a while. Are there other obvious
errors you're getting? Do other SVG's work?

dgtlmoon, 2018-11-02:
@jsphpl my two cents here, be careful your SVG's do not include groupings... could you possibly attach your SVG here for me to look at?

chilipeppr, 2018-11-02:
Yeah, the SVG interpreter is pretty lightweight, so you have to ungroup
everything in inkscape (or other app) first. The thing is, the sample
Chilipeppr SVG definitely was working, so it probably is some other error
that popped up recently.

jsphpl, 2018-11-02:
So here's everything printed to the console with log level 'error' when reloading on the jpadie workspace after clearing all application data:

```
GET http://chilipeppr.com/js/embed/highlight.pack.js?_=1541192399433 404 (Not Found)

VM284:1 Uncaught SyntaxError: Unexpected identifier
    at eval (<anonymous>)
    at Function.globalEval (jquery-2.1.0.min.js:2)
    at text script (jquery-2.1.0.min.js:4)
    at vc (jquery-2.1.0.min.js:4)
    at x (jquery-2.1.0.min.js:4)
    at XMLHttpRequest.<anonymous> (jquery-2.1.0.min.js:4)
    at Object.send (jquery-2.1.0.min.js:4)
    at Function.ajax (jquery-2.1.0.min.js:4)
    at Function.o._evalUrl (jquery-2.1.0.min.js:4)
    at o.fn.init.domManip (jquery-2.1.0.min.js:3)

GET http://chilipeppr.com/js/embed/embed.js?_=1541192399434 404 (Not Found)

VM285:1 Uncaught SyntaxError: Unexpected identifier
    at eval (<anonymous>)
    at Function.globalEval (jquery-2.1.0.min.js:2)
    at text script (jquery-2.1.0.min.js:4)
    at vc (jquery-2.1.0.min.js:4)
    at x (jquery-2.1.0.min.js:4)
    at XMLHttpRequest.<anonymous> (jquery-2.1.0.min.js:4)
    at Object.send (jquery-2.1.0.min.js:4)
    at Function.ajax (jquery-2.1.0.min.js:4)
    at Function.o._evalUrl (jquery-2.1.0.min.js:4)
    at o.fn.init.domManip (jquery-2.1.0.min.js:3)

GET http://chilipeppr.com/js/embed/highlight.pack.js?_=1541192399435 404 (Not Found)

VM288:1 Uncaught SyntaxError: Unexpected identifier
    at eval (<anonymous>)
    at Function.globalEval (jquery-2.1.0.min.js:2)
    at text script (jquery-2.1.0.min.js:4)
    at vc (jquery-2.1.0.min.js:4)
    at x (jquery-2.1.0.min.js:4)
    at XMLHttpRequest.<anonymous> (jquery-2.1.0.min.js:4)
    at Object.send (jquery-2.1.0.min.js:4)
    at Function.ajax (jquery-2.1.0.min.js:4)
    at Function.o._evalUrl (jquery-2.1.0.min.js:4)
    at o.fn.init.domManip (jquery-2.1.0.min.js:3)

GET http://chilipeppr.com/js/embed/embed.js?_=1541192399436 404 (Not Found)

VM289:1 Uncaught SyntaxError: Unexpected identifier
    at eval (<anonymous>)
    at Function.globalEval (jquery-2.1.0.min.js:2)
    at text script (jquery-2.1.0.min.js:4)
    at vc (jquery-2.1.0.min.js:4)
    at x (jquery-2.1.0.min.js:4)
    at XMLHttpRequest.<anonymous> (jquery-2.1.0.min.js:4)
    at Object.send (jquery-2.1.0.min.js:4)
    at Function.ajax (jquery-2.1.0.min.js:4)
    at Function.o._evalUrl (jquery-2.1.0.min.js:4)
    at o.fn.init.domManip (jquery-2.1.0.min.js:3)

Refused to apply style from 'http://chilipeppr.com/css/embed/embed-light.css' because its MIME type ('text/plain') is not a supported stylesheet MIME type, and strict MIME checking is enabled.

VM300:1253 WebSocket connection to 'ws://localhost:8989/ws' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED

GET https://googleads.g.doubleclick.net/pagead/id 0 ()

GET https://static.doubleclick.net/instream/ad_status.js 0 ()

GET chrome-extension://invalid/ 0 ()

GET chrome-extension://invalid/ 0 ()
[... 50 lines cut ...]
    at Object.onPubSubFileLoaded (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:1050:18)
    at publish (amplify.js:35)
    at Object.publish (app.js:167)
    at Object.loadGcode (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:190:28)
    at Object.loadGcodeDoNotCreateRecentFileEntry (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:162:18)
    at publish (amplify.js:35)
```

And again after choosing "mm" as the unit for the import:

```
VM301:1806 Uncaught TypeError: Cannot read property 'toFixed' of undefined
    at Object.decorateExtents (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:1806:58)
    at Object.drawExtentsLabels (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:2123:18)
    at Object.drawAxesToolAndExtents (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:1918:18)
    at Object.openGCodeFromText (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:1119:18)
    at Object.onPubSubFileLoaded (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:1050:18)
    at publish (amplify.js:35)
    at Object.publish (app.js:167)
    at Object.loadGcode (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:190:28)
    at Object.loadGcodeDoNotCreateRecentFileEntry (eval at globalEval (jquery-2.1.0.min.js:2), <anonymous>:162:18)
    at publish (amplify.js:35)
```

Repeating the entire procedure with adblocker disabled yields the same result.

jsphpl, 2018-11-02:
> Are there other obvious errors you're getting?

Yes:

[image]

jsphpl, 2018-11-02:
Loading an SVG file with the following content also results in the `toFixed` error:

```svg
<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" width="200" height="200">
    <rect width="200" height="200" style="stroke:#000000;stroke-width:1" />
</svg>

```

chilipeppr, 2018-11-03:
Yeah, something in that screenshot isn't right. That does indicate a
JavaScript error is happening on load of your page. Have you tried a
different browser? Or, what browser are you using?

jsphpl, 2018-11-03:
The above results stem from elston-chromium. Same symptoms (graphical glitches and toFixed error) happen in Firefox 63. Both on Mac.

dgtlmoon, 2018-11-03:
@chilipeppr looks like a script is missing `GET http://chilipeppr.com/js/embed/highlight.pack.js?_=1541192399433 404 (Not Found)`

jsphpl, 2018-11-04:
Used this button and localstorage is empty afterwards:

[image]

dgtlmoon, 2018-11-04:
@jsphpl so what was the outcome after you cleared it then?

jsphpl, 2018-11-04:
Everything i posted above happened with completely cleared storage