chilipeppr/tinyg #41: Buffer Overflow pausing TinyG2 / TinyG2 not executing commands

usernameusernamed, 2016-12-23:
I am receiving no motor movement. It appears that the Due that TinyG2 is running on is not executing commands. The buffer then overflows and the controller goes into pause mode.

In Chilipeppr after selecting the port it completes all commands (the check turns black) except {"sr":{"line":t,"posx":t,"posy":t,"posz":t,"vel":t,"unit":t,"stat":t,"feed":t,"coor":t,"momo":t,"plan":t,"path":t,"dist":t,"mpox":t,"mpoy":t,"mpoz":t}} (the check-mark for this command remains yellow).

Any ideas?

Here's information from the cmd running serial-port-json-server.exe -v : (verbose is on)

2016/12/22 17:51:34 seriallist_windows.go:231: index:0, name:COM7, friendly:USB Serial Device (COM7)
2016/12/22 17:51:34 seriallist_windows.go:232: pid:606D, vid:1D50
2016/12/22 17:51:34 seriallist_windows.go:239: Found related element1:{COM7 USB Serial Device (COM7) [] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}, element2:{COM8 USB Serial Device (COM8) [] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}
2016/12/22 17:51:34 seriallist_windows.go:231: index:1, name:COM8, friendly:USB Serial Device (COM8)
2016/12/22 17:51:34 seriallist_windows.go:232: pid:606D, vid:1D50
2016/12/22 17:51:34 seriallist_windows.go:239: Found related element1:{COM8 USB Serial Device (COM8) [] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}, element2:{COM7 USB Serial Device (COM7) [COM8] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}
2016/12/22 17:51:34 serial.go:123: Inside run of serialhub
2016/12/22 17:51:34 cayenn.go:62: UDP server running on port 8988 to listen for incoming device announcements.
2016/12/22 17:51:34 main.go:144: Your serial ports:
2016/12/22 17:51:34 main.go:151:        {COM7 USB Serial Device (COM7) [COM8] 2C58AE1D  Microsoft  606D 1D50}
2016/12/22 17:51:34 main.go:151:        {COM8 USB Serial Device (COM8) [COM7] 2C58AE1D  Microsoft  606D 1D50}
2016/12/22 17:51:54 conn.go:45: Started a new websocket handler
2016/12/22 17:51:55 hub.go:122: list
2016/12/22 17:51:58 seriallist_windows.go:198: DeviceId elements:[USB\VID_1D50 PID_606D MI_00\6 2C58AE1D 0 0000]
2016/12/22 17:51:58 seriallist_windows.go:198: DeviceId elements:[USB\VID_1D50 PID_606D MI_02\6 2C58AE1D 0 0002]
2016/12/22 17:51:58 seriallist_windows.go:231: index:0, name:COM7, friendly:USB Serial Device (COM7)
2016/12/22 17:51:58 seriallist_windows.go:232: pid:606D, vid:1D50
2016/12/22 17:51:58 seriallist_windows.go:239: Found related element1:{COM7 USB Serial Device (COM7) [] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}, element2:{COM8 USB Serial Device (COM8) [] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}
2016/12/22 17:51:58 seriallist_windows.go:231: index:1, name:COM8, friendly:USB Serial Device (COM8)
2016/12/22 17:51:58 seriallist_windows.go:232: pid:606D, vid:1D50
2016/12/22 17:51:58 seriallist_windows.go:239: Found related element1:{COM8 USB Serial Device (COM8) [] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}, element2:{COM7 USB Serial Device (COM7) [COM8] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}
2016/12/22 17:51:58 serial.go:489: Got metadata on ports:[{COM7 USB Serial Device (COM7) [COM8] 2C58AE1D  Microsoft USB Serial Device 606D 1D50} {COM8 USB Serial Device (COM8) [COM7] 2C58AE1D  Microsoft USB Serial Device 606D 1D50}]
2016/12/22 17:51:58 serial.go:546: About to marshal the serial port list. spl:{[{COM7 USB Serial Device (COM7) 2C58AE1D  false false [COM8] 0  [default timed nodemcu tinyg tinyg_old tinyg_linemode tinyg_tidmode tinygg2 grbl marlin] 1.88 1D50 606D 0} {COM8 USB Serial Device (COM8) 2C58AE1D  false false [COM7] 0  [default timed nodemcu tinyg tinyg_old tinyg_linemode tinyg_tidmode tinygg2 grbl marlin] 1.88 1D50 606D 0}]}
2016/12/22 17:52:00 hub.go:122: fro COM7
2016/12/22 17:52:00 feedrateoverride.go:34: Inside spFeedRateOverride arg: fro COM7 \n
2016/12/22 17:52:00 feedrateoverride.go:39: [fro COM7]
2016/12/22 17:52:00 feedrateoverride.go:48: The port to write to is:COM7---
2016/12/22 17:52:00 serial.go:606: Sending err back:  We could not find the serial port COM7 that you were trying to apply the feedrate override to.
2016/12/22 17:52:10 hub.go:122: close COM7
2016/12/22 17:52:10 hub.go:167: About to split close commands. cmd:"close COM7"
2016/12/22 17:52:10 hub.go:171: The split args for close:[close COM7]
2016/12/22 17:52:10 serial.go:606: Sending err back:  We could not find the serial port COM7 that you were trying to close.
2016/12/22 17:52:13 hub.go:122: open COM7 115200 tinygg2
2016/12/22 17:52:13 serialport.go:373: Inside spHandler
2016/12/22 17:52:13 serialport.go:388: Opening serial port COM7 at 115200 baud
2016/12/22 17:52:13 serial_windows.go:207: Byte val of commstat flags[0]: 10001
2016/12/22 17:52:13 serial_windows.go:208: Byte val of commstat flags[1]: 10000
2016/12/22 17:52:13 serialport.go:433: Just tried to open port
2016/12/22 17:52:13 serialport.go:442: Opened port successfully
2016/12/22 17:52:13 serial.go:132: Registering a port: COM7
2016/12/22 17:52:13 bufferflow_tinygg2.go:309: < {"r":{"fv":0.970,"fb":78.02,"cv":5,"hp":3,"hv":0,"id":"02130215d43","msg":"SYSTEM READY"},"f":[1,0,0]}
2016/12/22 17:52:13 bufferflow_tinygg2.go:435: We should RARELY get here cuz we should have a command in the queue to dequeue when we get the r:{} response. If you see this debug stmt this is one of those few instances where TinyG sent us a r:{} not in response to a command we sent.
2016/12/22 17:52:13 bufferflow_tinygg2.go:309: < tinyg [mm] ok>
2016/12/22 17:52:13 hub.go:122: list
2016/12/22 17:52:13 serial.go:252: Done with writeJson method
2016/12/22 17:52:13 serialport.go:233: Got p.sendBuffered. data:{"rxm":0}\n, id:internalInit0, pause:0
2016/12/22 17:52:13 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:{"rxm":0}
, id:internalInit0) start
2016/12/22 17:52:13 bufferflow_tinygg2.go:225: New line length: 10, buffer size increased to:10
2016/12/22 17:52:13 serialport.go:268: Got p.sendNoBuf. id:internalInit0, pause:0, data:{"rxm":0}\n
2016/12/22 17:52:13 serialport.go:275: Items In SPJS Queue List:0
2016/12/22 17:52:13 bufferflow_tinygg2.go:309: < {"r":{"rxm":null},"f":[1,100,9]}
2016/12/22 17:52:13 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:0, lenOfBuf:0
2016/12/22 17:52:13 seriallist_windows.go:120: Got an error back from oleutil.CallMethod. err:Exception occurred.
2016/12/22 17:52:13 serial.go:489: Got metadata on ports:[]
2016/12/22 17:52:13 serial.go:546: About to marshal the serial port list. spl:{[{COM7 COM7   true true [] 115200 tinygg2 [default timed nodemcu tinyg tinyg_old tinyg_linemode tinyg_tidmode tinygg2 grbl marlin] 1.88   0} {COM8 COM8   false false [] 0  [default timed nodemcu tinyg tinyg_old tinyg_linemode tinyg_tidmode tinygg2 grbl marlin] 1.88   0}]}
2016/12/22 17:52:13 bufferflow_tinygg2.go:309: < {"sr":{"line":0,"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"momo":4,"stat":1}}
2016/12/22 17:52:15 hub.go:122: sendjson {"P":"COM7","Data":[{"D":"?\n","Id":"tinygInit-cmd0","Pause":150},{"D":"{\"js\":1}\n","Id":"tinygInit-cmd1","Pause":150},{"D":"{\"sr\":n}\n","Id":"tinygInit-cmd2","Pause":0},{"D":"{\"sv\":1}\n","Id":"tinygInit-cmd3","Pause":50},{"D":"{\"si\":250}\n","Id":"tinygInit-cmd4","Pause":50},{"D":"{\"qr\":n}\n","Id":"tinygInit-cmd5","Pause":0},{"D":"{\"qv\":1}\n","Id":"tinygInit-cmd6","Pause":50},{"D":"{\"ec\":0}\n","Id":"tinygInit-cmd7","Pause":50},{"D":"{\"jv\":4}\n","Id":"tinygInit-cmd8","Pause":50},{"D":"{\"hp\":n}\n","Id":"tinygInit-cmd9","Pause":0},{"D":"{\"fb\":n}\n","Id":"tinygInit-cmd10","Pause":0},{"D":"{\"mt\":n}\n","Id":"tinygInit-cmd11","Pause":0},{"D":"{\"sr\":{\"line\":t,\"posx\":t,\"posy\":t,\"posz\":t,\"vel\":t,\"unit\":t,\"stat\":t,\"feed\":t,\"coor\":t,\"momo\":t,\"plan\":t,\"path\":t,\"dist\":t,\"mpox\":t,\"mpoy\":t,\"mpoz\":t}}\n","Id":"tinygInit-cmd12","Pause":250}]}
2016/12/22 17:52:15 serial.go:252: Done with writeJson method
2016/12/22 17:52:15 serialport.go:233: Got p.sendBuffered. data:?\n, id:tinygInit-cmd0, pause:150
2016/12/22 17:52:15 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:?
, id:tinygInit-cmd0) start
[... 108 lines cut ...]
2016/12/22 17:52:16 bufferflow_tinygg2.go:192: BlockUntilReady(cmd:{"mt":n}
, id:tinygInit-cmd11) start
2016/12/22 17:52:16 bufferflow_tinygg2.go:225: New line length: 9, buffer size increased to:27
2016/12/22 17:52:16 serialport.go:268: Got p.sendNoBuf. id:tinygInit-cmd10, pause:0, data:{"fb":n}\n
2016/12/22 17:52:16 serialport.go:275: Items In SPJS Queue List:2
[... cut ...]
2016/12/22 17:52:16 bufferflow_tinygg2.go:345: Buffer decreased to itemCnt:1, lenOfBuf:152
2016/12/22 17:52:16 hub.go:122: fro COM7
2016/12/22 17:52:16 feedrateoverride.go:34: Inside spFeedRateOverride arg: fro COM7 \n
2016/12/22 17:52:16 feedrateoverride.go:39: [fro COM7]
2016/12/22 17:52:16 feedrateoverride.go:48: The port to write to is:COM7---
2016/12/22 17:52:16 hub.go:122: sendjson {"P":"COM7","Data":[{"Id":"startup0","D":null}]}
2016/12/22 17:52:16 serial.go:252: Done with writeJson method

chilipeppr, 2016-12-23:
Did you turn on "No Init Cmds" in the tinyg widget upper right corner
pulldown menu? Different versions of Due firmware either seem to support
the TinyG v8 commands, or they crap out. It's cuz you're on v9 which is in
flux.

usernameusernamed, 2017-01-08:
This works!
Turned "No Init Cmds" on and all is running smoothly now.

Thank you

closed 2017-01-08 by usernameusernamed