chilipeppr/tinyg #44: chilipeppr.com does not work

m0rt3nlund, 2017-10-31:
Hi!

The page does not currently work. It only replies with "Edit in JSFiddle"

Morten

m0rt3nlund, 2017-10-31:
I suspect that theres is missing some directories on the server as the directory "embed" does not exist anymore?

chilipeppr, 2017-10-31:
Oh boy. Working on it.

m0rt3nlund, 2017-10-31:
Now the page works, but there is still some "Edit in JSFiddle" here and there :-) The GCode module etc has this over the list of gcodes.

Morten

m0rt3nlund, 2017-11-01:
Hi again!

There is still some issues i guess.. Lots of 404 for misc files.

Added the trace from the console in the file below!
Hope you can fix this as I really enjoy using Chilipeppr!

[chilipeppr.com-1509563924789.txt](https://github.com/chilipeppr/tinyg/files/1435390/chilipeppr.com-1509563924789.txt)

chilipeppr, 2017-11-01:
what workspace are you using?

m0rt3nlund, 2017-11-01:
I am currently trying to use grbl!
I noticed now when I tried the tinyG workspace that everything worked better there, except when you choose the "Workspaces" button to see the list of workspaces, then I get another one of these "Edit in JSFiddle" messages.

chilipeppr, 2017-11-01:
Yeah, the tinyg got all fixed yesterday. There were 5 widgets I had to move
to Github super fast and that solved it. The Your Workspaces widget is
still JSFiddle, so that is an open issue. The problem with Grbl workspace
is almost every key widget was still in JSFiddle. Would you be up for
porting them to Github real fast? I can add you to the Github repo for Grbl
as an admin so you have access to all of it.

m0rt3nlund, 2017-11-01:
Oh, I would really like to help with that, but I really would not know how to do that I am afraid! :-/

chilipeppr, 2017-11-01:
If I show you how to do one, would you be up for doing the other 6 or so?
It's literally creating a new Github repo for the widget, importing the
widget template to the repo, cutting pasting the Javascript, HTML, and CSS
from JSFiddle into the template, pushing back to Github, then re-pointing
the workspace Javascript to the Github repo instead of the JSFiddle. It's
kind of repetitive.

m0rt3nlund, 2017-11-01:
Yes of course! Just give me some directions and i will execute! :) I am familiar with git at least :-)

chilipeppr, 2017-11-02:
Here's a video I made last night that shows how to fix a widget.
https://www.youtube.com/watch?v=KX9B4rtfbyo