chilipeppr/tinyg #46: Does not respond to commands of any type

therobotguyllc, 2019-12-09:
Greetings,

I am trying to migrate from my old Uno GRBL setup to a Tinyg2/Due setup. I have tried both UGS and Chilipeppr and they both connect to the board but neither one can get any action out of it. I can't jog at all. When I run gcode the serial terminal turns into a matrix screen saver but nothing happens. In UGS it just say "OFF-LINE" all the time. I don't really know what to do to troubleshoot this. I loaded TinyG from the bin file using Chilipeppr. Maybe I did something wrong? Can't seem to find any answers. 

Any direction would be helpful.

thanks,

therobotguyllc, 2019-12-09:
Update:
After a number of frustration I finally decided it might just be a driver issue. I rebooted by computer and also the Due board and now it works with both UGS and Chilipeppr. Not entirely sure what the issue was but it seems to be fixed now. It appears like my PC serial driver was not working correctly.

closed 2019-12-09 by therobotguyllc

chilipeppr, 2019-12-09:
Awesome. Glad to hear you got it figured out.