chilipeppr/tinyg #48: Internal Server Error  - Chilipeppr/Tinyg not opening on web browser

GravityInc, 2020-08-05:
Hi Guys,

Hope you all are safe and doing well....

Today i tried to launch chilipeppr/tinyg on my browser and it was loading a blank page.

When i inspect the webpage i see the below. Please can you suggest me a way to resolve it.

[image]

chilipeppr, 2020-08-05:
Google shutdown anything below go 1.10 for app engineer and thus chilipeppr
needs recompiled, but the problem is they deprecated core libraries so
certain things need reimplemented from scratch.

chilipeppr, 2020-08-06:
Ok. Fixed. Should be ok now.

Had to hard code all the popular workspaces to get this up and running quickly. The database model has to get rewritten this weekend, which is where the workspace boot scripts are stored. No logins work either, although there's very limited functionality that relies on that, so we should be fine now for everyone to be functioning again.

GravityInc, 2020-08-07:
Thanks mate! really appreciate you effort and this application.

Was wondering if there is a way I can contribute to the code?

I will close this issue....

closed 2020-08-07 by GravityInc