chilipeppr/tinyg #49: chilipeppr.com doesn't work#

navinvelappan, 2020-08-05:
Chilipeppr doesn't opens up in any browser. Just a blank white page loads .

chilipeppr, 2020-08-05:
Google shutdown anything below go 1.10 for app engine and thus chilipeppr
needs recompiled, but the problem is they deprecated core libraries so
certain things need reimplemented from scratch.

chilipeppr, 2020-08-05:
Working on it. Could take 24 hours.

navinvelappan, 2020-08-06:
It says URL is available and edit boot javascript. DOes that mean the chilipeppr is available. if yes how to do that edit boot javascript.

chilipeppr, 2020-08-06:
No, it's only half fixed. Making progress. Stay tuned...

chilipeppr, 2020-08-06:
Ok. Fixed. Should be ok now.

Had to hard code all the popular workspaces to get this up and running quickly. The database model has to get rewritten this weekend, which is where the workspace boot scripts are stored. No logins work either, although there's very limited functionality that relies on that, so we should be fine now for everyone to be functioning again.