chilipeppr/tinyg #50: Chilipeppr and JSON 1.94/1.96 connection issues

CorbPeters, 2020-09-22:
I was using chilipeppr one day on JSON 1.94 and then the next day it stopped working smoothly. I have a linux machine running JSON and a windows machine run chilipeppr with Chrome. I have had the linux machine running for months without having to restart JSON and then all the sudden it stops connecting. Since then every time I want to run a job I have to unplug the G2, restart JSON, and then bring up chilipeppr to be able to connect. Then some of the buttons like the mm/in button may or may not work. If I want to run multiple jobs I have to close chilipeppr, reset the g2, restart json and then restart chilipeppr, change mm to in using the G20 command since the button doesn't work, then I can load my project. I updated to JSON 1.96 and it didn't seem to help.

Seems like something updated in the background between day 1 and day 2 in chilipeppr or chrome maybe?

CorbPeters, 2020-09-22:
May have been an issue on my end.

closed 2020-09-22 by CorbPeters

chilipeppr, 2020-09-22:
Was it software? Or was it a USB cable?

CorbPeters, 2020-09-22:
The hard drive in the laptop was loose so when I would put it down the PC would turn off sometimes as the hard drive disconnected. PEDKAC/id10t error. No idea why the other stuff started working ok but today it seems to be working. No issues with closing and reloading chilipeppr and JSON disconnects. I can reset the G2 from chilipeppr again without having to restart JSON which is needed because I can't seem to make the G2 switch jobs. If I start a job and then it finishes or I stop it and I drag a new job in and click play it just runs the old job again even if I flush the queue. I am running a modified 100.26 firmware though so maybe this changed?

chilipeppr, 2020-09-22:
Hmm. The same job running may be ChiliPeppr. I've heard folks say that can
sometimes happen. I recommend you wipe all the old files to ensure the
browser has lots of room to load the new Gcode file.