chilipeppr/tinyg #51: Chillipeppr.com not working?

Youriyj, 2021-02-02:
Hi the website doesnt seem to be working is there anything wrong?

chilipeppr, 2021-02-02:
It's working fine. You sure you don't have an Ad Blocker turned on? The
site uses Google Analytics and I've heard folks with ad blockers get a
white screen.

Youriyj, 2021-02-02:
I just tried it and that was the issue thanks for the fast reply. Also i see you only designed the robot actuators for the wrists have you made the ones for the base and other ones aswell or is that still in the making?

chilipeppr, 2021-02-02:
I tried to move to a brushless motor to gain more power and that caused a
full rewrite/redesign of almost everything. So, I doubt I'll get further on
this project.

Youriyj, 2021-02-02:
Ah that’s unfortunate ill have to design something myself. Do you know how much load this arm can hold, im just making this for fun and learning experience and want to use it to move objects under 2kg