chilipeppr/tinyg #52: CNC Machine Moving in Wrong Direction

jneterer, 2021-08-05:
My grandfather recently bought a CNC machine (using TinyG) that was built by another individual. The builder demoed it to him when he bought it and it was working. When my grandfather took it home and ran his first project, the router was moving in wrong directions, as if it was programmed wrong. For example, if he wanted it to move `+x`, it might move `+z`, or if he wanted it to move `+y`, it might move `-y-z`. He called the original builder and he was able to remote into his laptop (Windows) and fix it, but he's not sure how he fixed it.

Recently I took the laptop home with me to design a sign and when I brought it back and plugged it into the CNC machine, it's now behaving the same way, and we can't get ahold of the original owner to fix it. I have a CS degree (I'm a software engineer) so I know my way around these things, but I've never worked with CNC software before so I'm a little out of my element. I'm hoping you can give me some direction and I might be able to figure it out. For example, do you think something is off with TinyG? Or is there some kind of configuration in Chilipeppr that I need to make?

Thanks for your help! @chilipeppr

chilipeppr, 2021-08-05:
This feels like something with the configuration of the TinyG. When you run
any Gcode, do you get consistent behavior? If a +x always gives you a +z
then it really is just the config.

Quas7, 2021-08-05:
I would assume, the wiring is just mixed up after transport. Just swap the connections ad required (X vs Z). If directions are wrong afterwards, this can be handled in software. But depending on the machine, the coords are tool centered. Do not confuse movment of table with movment of tool (as I once did).

jneterer, 2021-08-05:
@chilipeppr It sounds like the config of TinyG then, because movements have consistent incorrect behavior.

@Quas7 I would consider the possibility of wiring being mixed up, however the original builder of the machine was able to fix the issue by using a remote connection to the laptop and fixing it, so nothing was changed hardware-wise.

jneterer, 2021-08-05:
@chilipeppr being that the config of TinyG is wrong, do you know of any resources I could reference to fix it? Or a place I could get some help?

Quas7, 2021-08-05:
https://github.com/synthetos/TinyG/wiki/TinyG-Configuration